kantlivelong/OctoPrint-SmartPreheat #1: Cannot save G-code template changes (js error: "Error calling onSettingsBeforeSave on view model")

dlitz, 2020-05-02:
#### What were you doing?

1. Install Smart Preheat plugin via Plugin Manager
2. Open OctoPrint Settings -> Plugin -> Smart Preheat
3. Modify the g-code template
4. Click "Save"
5. Open OctoPrint Settings -> Plugin -> Smart Preheat again

#### What did you expect to happen?

- Modified G-code should appear in the text area.
- Modified G-code should appear in ~/.octoprint/scripts/gcode/snippets/doSmartPreheat

#### What happened instead?

- Previous G-code appears in the text area.  Changes are lost.
- ~/.octoprint/scripts/gcode/snippets/doSmartPreheat is not modified
- No network request was made from the browser (Firefox 75.0 on Linux/x86_64)
- Error in JavaScript console:

```
Error calling onSettingsBeforeSave on view model SmartPreheatViewModel : SmartPreheatViewModel/self.onSettingsBeforeSave@https://octoprint.dlitz.net/static/webassets/packed_plugins.js?02445d5c:10054:25
callViewModelIf@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:16543:34
callViewModelsIf/<@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:16489:28
Pn@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:11208:530
ur/<@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:11229:66
callViewModelsIf@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:16487:7
callViewModels@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:16481:21
SettingsViewModel/self.onAllBound/<@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:5306:31
dispatch@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:5184:27
add/elemData.handle@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:4992:28
trigger@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:8250:12
trigger/<@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:8328:17
each@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:355:19
each@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:190:17
trigger@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:8327:15
SettingsViewModel/self.saveData@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:5731:33
click/<@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949 line 11374 > Function:3:85
init/</<@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:11395:443
dispatch@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:5184:27
add/elemData.handle@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:4992:28
packed_core.js:16553:17

Error calling onSettingsHidden on view model SmartPreheatViewModel : SmartPreheatViewModel/self.onSettingsHidden@https://octoprint.dlitz.net/static/webassets/packed_plugins.js?02445d5c:10049:21
callViewModelIf@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:16543:34
callViewModelsIf/<@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:16489:28
Pn@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:11208:530
ur/<@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:11229:66
callViewModelsIf@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:16487:7
callViewModels@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:16481:21
SettingsViewModel/self.onAllBound/<@https://octoprint.dlitz.net/static/webassets/packed_core.js?3514a079:5302:35
dispatch@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:5184:27
add/elemData.handle@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:4992:28
trigger@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:8250:12
trigger/<@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:8328:17
each@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:355:19
each@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:190:17
trigger@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:8327:15
hideModal@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:14740:6
hideWithTransition/<@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:14724:10
fn@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:4933:18
dispatch@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:5184:27
add/elemData.handle@https://octoprint.dlitz.net/static/webassets/packed_libs.js?3339d949:4992:28
packed_core.js:16553:17
```

#### Version of OctoPrint-SmartPreheat

0.0.1

#### Operating System running OctoPrint

OctoPi 0.15.1

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.smartpreheat set to DEBUG

- [octoprint.log](https://github.com/kantlivelong/OctoPrint-SmartPreheat/files/4568512/octoprint.log)

kantlivelong, 2020-05-02:
Guess I broke it when cleaning up the code before submission. Would you mind installing [this](https://github.com/kantlivelong/OctoPrint-SmartPreheat/archive/745846f7a509ac772007714d77681dc8ead4b952.zip) patched version and let me know if it works for you?

dlitz, 2020-05-03:
Works great!

kantlivelong, 2020-05-03:
Pushed the release. Thanks for your help!

closed 2020-05-03 by kantlivelong