kantlivelong/OctoPrint-SmartPreheat #3: Installloop to 0.0.2

mindfest, 2020-05-03:
#### What were you doing?

1. Installed SmartPreheat through Octopi plugin manager
2. Restarted the server
3. Confirmed Updatemessage to Version 0.0.2
4. Restarted the server
5. Another update mesage appeared
6. Confirmed again to see what happens
7. Then i saw in the message box that only version .0.01 was installed again through the updater
8. After restart the update message was there again
9. Uninstalled SmartPreheat plugin

#### Version of OctoPrint-SmartPreheat
0.0.1

#### Operating System running OctoPrint
Octopi 1.4.0 on Raspberry Pi 3b

#### Printer model & used firmware incl. version
Anet A8, Merlin 2.0.5.3

ojaksch, 2020-05-03:
Same here. Took a look at the package and found that setup.py identifies itself as **plugin_version = "0.0.1"**

buzzlaker, 2020-05-03:
Confirmed as well.

nikcoultas, 2020-05-04:
Yup, same here.

matburnham, 2020-05-04:
[image]

ptwats, 2020-05-05:
Here is a log file if that helps.
[plugin_pluginmanager_console.log.txt](https://github.com/kantlivelong/OctoPrint-SmartPreheat/files/4580948/plugin_pluginmanager_console.log.txt)

kantlivelong, 2020-05-05:
Apparently this repo got marked as unwatched for me. Doh!

I forgot to bump the version in setup.py but 0.0.3 was just released which fixes a parsing issue. Give it a go and let me know.

matburnham, 2020-05-05:
Works for me. 👍

kantlivelong, 2020-05-05:
Marking as solved then.

Apologies for the delay, had no idea I even had tickets open.

closed 2020-05-05 by kantlivelong