kantlivelong/OctoPrint-SmartPreheat #4: Hanging if you just print

LMS0815, 2020-05-04:
If there is only one file in file manager and you just hit print button, the plugin seems hanging. OctoPrint needs to be rebooted.
I assume, the variables _plugins.smartpreheat.bed_  and/or are _plugins.smartpreheat.tools.items()_ not set.  In addition it might be worth to amend the script...

```
{% if printer_profile.heatedBed and plugins.smartpreheat.bed %}
        M190 S{{ (plugins.smartpreheat.bed * 0.8)|round }}
        M140 S{{ plugins.smartpreheat.bed }}
{% endif %}

```
I do not know if the **FILE_SELECTED** is triggered if you just print.
You probably need to add an event like **PRINT_STARTED** as well to trigger value initialization.

kantlivelong, 2020-05-05:
A few bugs with the parser were just fixed in 0.0.3 which may have resulted in temps set later in the gcode file to be used. 

As far as the click print with one file thing, I don't have the ability to click print without selecting a file. Could you update first and try again with logging?

yanfali, 2020-05-06:
Hey, awesome idea for a plugin, but I had to uninstall it. When I loaded large gcode files into my octoprint instance, it would basically cause the server to hang while the plugin went off and ran regex against a 15MB gcode file. It took me a couple of days to figure out a pattern. Sure enough, uninstalling it fixed my stability issues. This actually ruined a print as octoprint completely crashed.

Thanks again, I really like the idea. Just wanted to let you know what I experienced.

kantlivelong, 2020-05-06:
@yanfali There was an issue fixed with the whole file being read when it shouldn't have been. 0.0.3 released today fixing it. Give it another shot.

LMS0815, 2020-05-06:
The hanging issue is related to the system (windows/linux) where I installed this plugin.
Its probably related to the snippetas the revised code is only reading some lines.

You can print a file without selecting it, if it was printed befor by just ckicking the **[Print]** button.
[image]

kantlivelong, 2020-05-06:
@LMS0815 I've pushed 94b9714a8de2ef6df347729448a4e26c256b5f3b which reworks the scan trigger. Should be much more reliable.

LMS0815, 2020-05-07:
As I am using the forked version now the issue is gone. 
I need to revert to your version which this will take a while.

Just check when the events are triggered by creating a log entry.
You will see, that SELECTED is triggered during _OctoPrint_ load which blocks loading if the plugin is hanging (=slow). Seems your new version is better taking care about long files now.

Just to be clear: I really appreciate your idea and work you shared with this plugin.
I was already planning to write my own as it is essential for me to preheat prior to send `G29 J;  Get Unified Bed Leveling Grid` prior to a print..

Just assume that there are _gcode_ files around which might not meet your assumptions and do not deliver the expected scan result.

**Therefor I recommend:**
1. Use PRINT_STARTED to avoid scan during _OctoPrint_ start.
2. Assume that reading 1000 lines of _gcode_ should be enough for scanning if the file is OK. Just stop.
3. Always Provide default values (GUI configuration - could be 0 temperature) if you can not scan or not scanning by purposes as any snippet needs to have initializes values otherwise it will crash.

closed 2020-05-07 by LMS0815