kantlivelong/OctoPrint-SmartPreheat #6: Improvement: Default values & Undefined tool

LMS0815, 2020-05-05:
My gcode files do not provide the tool ID. Please consider searching for tool set gcode without `T` parameter

```
M104 Sxx
M109 Sxx
```

I recommend adding tools ID **-1** to cover undefined = last use tool to work with the snippet below.
This code is in addition providing default values, if the according variables are not set.

```
; Initialize vars
 {%- set bed, list = plugins.smartpreheat.bed|default(80 , true), plugins.smartpreheat.tools|default({-1: 195} , true) %}

; Set bed
{%- if printer_profile.heatedBed %} 
M190 S{{ (bed * 0.8)|round }} ;Wait for Bed
M140 S{{ bed }} ;Set for Bed
{%- endif %}

; Set tool temps
{%- for tool, temp in list.items() %}
M104{{- '' if tool < 0 else ' T' + tool|string }} S{{ temp }} ;Set Hotend
{%- endfor %}

; Wait bed
{%- if printer_profile.heatedBed %} 
M190 S{{ bed }} ;Wait for Bed
{%- endif %}

; Wait tool temps
{%- for tool, temp in list.items() %}
M109{{- '' if tool < 0 else ' T' + tool|string }} S{{ temp }} ;Set Hotend
{%- endfor %}
```

Feel free to use it. Might be nice to be mentioned as the source :)

kantlivelong, 2020-05-05:
Apparently this repo got marked as unwatched for me. Doh!

A few bugs with the parser were just fixed in 0.0.3 which may have resulted in temps set later in the gcode file to be used.

Can you upgrade and try again. Please provide logs as detailed in the ticket submission process.

LMS0815, 2020-05-07:
Kindly always provide valid default values (=temperatures), even if no scanning occurred.

closed 2020-05-07 by LMS0815