kantlivelong/OctoPrint-SmartPreheat #11: it does not seem to be working for me.

leoheck, 2020-05-20:
Hi, it does not seem to be working for me.
I installed the plugin, but the graph looks pretty standard.

Should I configure the plugin before? Or it is ready for use?

[image]

kantlivelong, 2020-05-20:
You clearly didn't read [this](https://raw.githubusercontent.com/kantlivelong/OctoPrint-SmartPreheat/master/.github/ISSUE_TEMPLATE.md).

However [this](https://community.octoprint.org/t/smartpreheat-not-preheating-both/19476/16) may help you. If not then please do the needful.

leoheck, 2020-05-21:
Of course I didn't read a bunch of unecessary text. I am not demaning a fix. I am given you my feedback. But I agree with you, I will do the needful by uninstalling this.

closed 2020-05-21 by leoheck