kantlivelong/OctoPrint-SmartPreheat #12: Feature request: Detect when bed is already hot and start heating nozzle immediately.

Silverfire9, 2020-05-22:
Sorry about posting here, but the feature request link doesn't look like it's live yet. No entry for this plugin on the feathub/kantlivelong page.

I noticed that if I start another print immediately after finishing the last one, the plugin waits for the bed to cool down to the temperature it expects to change to both heaters starting before starting to heat the nozzle. I'm thinking it should be easy enough to add to the code so that it can detect whether the bed is in the range where both would be heating and just start everything.

Ex. Temperature settings are at 60*C and 200*C. Actual bed temperature is 54*C and nozzle temperature is 32*C

Current behaviour:
Plugin waits for bed to cool to 48*C before reheating the bed and nozzle at the same time.

Desired behaviour: 
Plugin detects that the bed is already warmer than the initial setting (but not warmer than the final setting to allow for different bed temperature settings ie. first print is ABS, at 110*C and second is PLA at 50*, but bed is still at 75*C) and start to heat both devices immediately.

Figured it would be as simple as a change from an '=' to a '>=' in one expression and the addition of a '<' expression elsewhere, but I don't "speak" js so didn't see where in the code.

kantlivelong, 2020-05-22:
Whoops. Thought I opened that up. Well it is now :smile: 

You can already do this by modifying the routine in settings. It's user customizable.

https://docs.octoprint.org/en/master/features/gcode_scripts.html#snippets

kantlivelong, 2020-05-22:
Though I'm also be curious about what your firmware is doing because `M190 S##` is a min temperature.

https://www.reprap.org/wiki/G-code#M190:_Wait_for_bed_temperature_to_reach_target_temp

Silverfire9, 2020-05-22:
That IS interesting. It is indeed enabled in the plugin settings. After a cursory glance at what M190 does and a quick google, it appears it's a Klipper fault per this report: https://github.com/KevinOConnor/klipper/issues/1108

While there are a number of good reasons I like Klipper over Marlin, the fact that its Marlin G-code compatibility is still under development does tend to cause problems. I don't have time to look into it further until tonight, but I will do so and get back to you if you would like with a proper solution. 

While I'm not quite to ready to close this without the Klipper solution, I think it's dealt with as much as is required for now since the solution doesn't seem to be with the plugin.

kantlivelong, 2020-05-22:
Good find. Really strange response from Klipper devs. What they seem to be doing is implementing `R` mode for both cases.

kantlivelong, 2020-05-22:
@Silverfire9 I just realized that [OctoPrint-SetAndWait](https://github.com/kantlivelong/OctoPrint-SetAndWait) may help you.

rashevskyv, 2021-03-24:
Did you made this snippet? 
@Silverfire9 can you share?

iqubik, 2022-06-26:
; Wait for bed to reach 80% of required temp then set to required temp
{% if printer_profile.heatedBed and last_temperature['b'].actual < (plugins.smartpreheat.bed * 0.9) %}
M190 S{{ (plugins.smartpreheat.bed * 0.9)|round }}
M140 S{{ plugins.smartpreheat.bed }}
{% else %}
M140 S{{ plugins.smartpreheat.bed }}
{% endif %}

; Set tool temps
{% for tool, temp in plugins.smartpreheat.tools.items() %}
M104 T{{ tool }} S{{ temp }}
{% endfor %}

; Wait for bed
{% if printer_profile.heatedBed %}
M190 S{{ plugins.smartpreheat.bed }}
{% endif %}

; Wait for tools
{% for tool, temp in plugins.smartpreheat.tools.items() %}
M109 T{{ tool }} S{{ temp }}
{% endfor %} 

This is solution

jezmck, 2022-10-09:
This is still an issue.