kantlivelong/OctoPrint-SmartPreheat #16: Plugin won't update to 0.0.6

NovaViper, 2021-02-02:
#### What were you doing?

1. Plugin says an update is available
2. I update the plugin and restart
3. Octoprint comes back, but plugin still says an update is availiable

#### What did you expect to happen?
It's suppose to update

#### What happened instead?
It appears it didn't actually update

#### Version of OctoPrint-SmartPreheat
0.0.6(?)

#### Operating System running OctoPrint
Octopi 0.17.0

#### Printer model & used firmware incl. version
Ender 3 Marlin 2.0.x bugfix

#### Link to octoprint.log with octoprnt.plugins.smartpreheat set to DEBUG
[octoprint.log](https://github.com/kantlivelong/OctoPrint-SmartPreheat/files/5912984/octoprint.log)

FlorianDessloch, 2021-02-02:
Same here.
I even removed 0.05 and installed 0.0.6 from scratch, it still says its 0.0.5 and there is an update.

kantlivelong, 2021-02-02:
Something got messed up with the tags. Try now.

NovaViper, 2021-02-02:
It's fixed now, thanks!

closed 2021-02-02 by NovaViper