kantlivelong/OctoPrint-SmartPreheat #17: Temperature offsets are not applied

richfelker, 2021-04-08:
When temperature offsets have been set via the OctoPrint temperature UI, Smart Preheat does not see them and preheats to the original temperatures in the sliced file rather than the offset ones. I looked in the OctoPrint documentation for a way to retrieve the offsets and apply them in the script but I could not find one. If such a method exists it would be nice to use it by default or at least have comments indicating how to enable using it. If not, maybe a method for exporting the value to gcode scripts could be added to Smart Preheat - I would assume the Python side has access to them in some way.

richfelker, 2021-04-08:
I was able to hack this in by adding `offsets = self._printer._comm.getOffsets()` and applying them to the value returned in `get_temps_from_file`.

kantlivelong, 2021-04-16:
To be honest I never tested w/ offsets and forgot it was even a feature. I'll look into supporting this.

richfelker, 2021-04-16:
Thanks. As for the motivation, I really need the starting temps the file to be printed will initially wait for (with offsets applied) because my "start print" gcode script in octoprint is responsible for priming. If there's any wait again between it and the file (to heat up from un-offset temp to offset temp) the nozzle will ooze, defeating the whole purpose of having primed it.

richfelker, 2022-04-05:
Ping. Proposed possible patch for this:

```
--- __init__.py.orig	2022-04-04 07:40:17.406609696 +0100
+++ __init__.py	2022-04-05 17:24:59.116629495 +0100
@@ -89,6 +89,8 @@
                 if gcode in ('M104', 'M109', 'M140', 'M190'):
                     self._logger.debug("Line %d: Detected SetTemp. Line=%s", lineNum, line)
 
+                    offsets = self._printer._comm.getOffsets()
+
                     toolMatch = octoprint.util.comm.regexes_parameters["intT"].search(line)
                     if toolMatch:
                         toolNum = int(toolMatch.group("value"))
@@ -101,10 +103,10 @@
 
                         if gcode in ("M104", "M109"):
                             self._logger.debug("Line %d: Tool %s = %s", lineNum, toolNum, temp)
-                            temps["tools"][toolNum] = temp
+                            temps["tools"][toolNum] = temp + offsets.get("tool%d" % toolNum, 0)
                         elif gcode in ("M140", "M190"):
                             self._logger.debug("Line %d: Bed = %s", lineNum, temp)
-                            temps["bed"] = temp
+                            temps["bed"] = temp + offsets.get("bed", 0)
 
         self._logger.debug("Temperatures: %r", temps)
         return temps
```