kantlivelong/OctoPrint-SmartPreheat #19: TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

guythnick, 2021-12-21:
I am getting an error in the logs that is causing prints to fail.  This was after upgrading OctoPrint to 1.7.2.  I made no other changes.  I also upgraded to Python 3 as well, but I am getting the same error.  For some reason it does not like my code snippet, which was not changed from when it was working before the upgrade. Also including that below. 

Octoprint - 1.7.2 Python3
OctoPrint-SmartPreheat - 0.6

```
2021-12-21 15:37:27,720 - octoprint.settings - ERROR - Exception while trying to render script gcode:beforePrintStarted
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/settings.py", line 1989, in loadScript
    script = template.render(**context)
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/.octoprint/scripts/gcode/beforePrintStarted", line 1, in top-level template code
    {% snippet 'doSmartPreheat' %}
  File "/home/pi/.octoprint/scripts/gcode/snippets/doSmartPreheat", line 4, in top-level template code
    M190 S{{ (plugins.smartpreheat.bed * 0.8)|round }}
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
```

[image]

LizardThief, 2023-08-16:
I think you need to replace the two `{% if printer_profile.heatedBed %}` with `{% if printer_profile.heatedBed and plugins.smartpreheat.bed != None %}`

I just encountered a similar error where a print job skipped my preheat altogether when I tried to print a file that had no bed heating commands

TheThief, 2023-08-16:
I think you need to replace the two `{% if printer_profile.heatedBed %}` with `{% if printer_profile.heatedBed and plugins.smartpreheat.bed != None %}`

I just encountered a similar error where a print job skipped my pre-print gcode (both preheating (expected) and homing and auto-leveling (not so expected) commands) when I tried to print a file that had no bed heating commands, so smart preheat couldn't get a bed temp