kantlivelong/OctoPrint-SmartPreheat #20: When actual extruder temp order is upper target order, decrease it immediatly

Alcolo47, 2023-07-24:
Concrete case:
With my Anycube cobra Neo, after "filament loading" the extruder temp order stay at 230.

When I start a print with extruder temp to 200, The plugin wait for bed intermediate temp before lowing extruder temp order.
But in this case is better to lowing extruder temp asap.

[image]

kantlivelong, 2023-07-24:
You can edit the gcode script to do this unless I'm missing something here.

Alcolo47, 2023-08-22:
In my example, you can see that I start printing with an hot tool (230).
I this case, don't wait for bed intermediate target, to decrease tool temp.

kantlivelong, 2023-08-22:
Again, this can be done in the snippet. This is considered configuration.

closed 2023-08-22 by kantlivelong